floatlazer/semantic_slam #2: bad callback

liyi2017, 2018-09-04:
hi,  i try your code have a problem, when i run the "roslaunch semantic_slam semantic_mapping.launch
", the error : 
[ERROR] [1536051546.135952]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7efd1010edd0>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 75, in callback
    self.signalMessage(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 57, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 287, in add
    self.signalMessage(*msgs)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 57, in signalMessage
    cb(*(msg + args))
  File "/home/l/catkin_ws/src/semantic_slam-master/semantic_cloud/src/semantic_cloud.py", line 198, in color_depth_callback
    depth_img = resize(depth_img, (self.img_height, self.img_width), order = 0, mode = 'reflect', anti_aliasing=False, preserve_range = True) # order = 0, nearest neighbour
TypeError: resize() got an unexpected keyword argument 'anti_aliasing'

i donot konw how to deal it , i need your heep ,thank you.

floatlazer, 2018-09-04:
Hi, I think that it is caused by the version of [skimage](https://scikit-image.org/download.html). I use it to resize images. Try to upgrade to version 0.14.0 should solve this error (Lower version resize methods are slightly different).

For example:

```sh
pip install --upgrade scikit-image
```

liyi2017, 2018-09-05:
yes, thank you very much.

closed 2018-09-05 by floatlazer

punnpkin, 2019-01-22:
hello there, how to solve the error when upgrading?
sudo pip install --upgrade scikit-image==0.14.0
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Collecting scikit-image==0.14.0
  Downloading https://pypi.tuna.tsinghua.edu.cn/packages/11/c7/ee75c79dcce057a3475763d611ec044737a708eaf5cc53426b0117795ddb/scikit_image-0.14.0-cp27-cp27mu-manylinux1_x86_64.whl (25.4MB)
    100% |████████████████████████████████| 25.4MB 14.4MB/s 
Collecting cloudpickle>=0.2.1 (from scikit-image==0.14.0)
  Downloading https://pypi.tuna.tsinghua.edu.cn/packages/fc/87/7b7ef3038b4783911e3fdecb5c566e3a817ce3e890e164fc174c088edb1e/cloudpickle-0.6.1-py2.py3-none-any.whl
Requirement already satisfied, skipping upgrade: pillow>=4.3.0 in ./.local/lib/python2.7/site-packages (from scikit-image==0.14.0) (5.4.0)
Requirement already satisfied, skipping upgrade: scipy>=0.17.0 in /usr/local/lib/python2.7/dist-packages (from scikit-image==0.14.0) (1.2.0)
Requirement already satisfied, skipping upgrade: six>=1.10.0 in ./.local/lib/python2.7/site-packages (from scikit-image==0.14.0) (1.12.0)
Requirement already satisfied, skipping upgrade: PyWavelets>=0.4.0 in /usr/local/lib/python2.7/dist-packages (from scikit-image==0.14.0) (1.0.1)
Requirement already satisfied, skipping upgrade: networkx>=1.8 in /usr/local/lib/python2.7/dist-packages (from scikit-image==0.14.0) (2.2)
Collecting matplotlib>=2.0.0 (from scikit-image==0.14.0)
  Downloading https://pypi.tuna.tsinghua.edu.cn/packages/89/0c/653aec68e9cfb775c4fbae8f71011206e5e7fe4d60fcf01ea1a9d3bc957f/matplotlib-3.0.2.tar.gz (36.5MB)
    100% |████████████████████████████████| 36.5MB 14.5MB/s 
    Complete output from command python setup.py egg_info:
    
    Matplotlib 3.0+ does not support Python 2.x, 3.0, 3.1, 3.2, 3.3, or 3.4.
    Beginning with Matplotlib 3.0, Python 3.5 and above is required.
    
    This may be due to an out of date pip.
    
    Make sure you have pip >= 9.0.1.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-4JndzI/matplotlib/

punnpkin, 2019-01-22:
i'm using python2.7

jsdd25, 2019-02-12:
@punnpkin ，Hi, I have the same problem as you. Did you fix it?
Thanks!

punnpkin, 2019-02-15:
> @punnpkin ，Hi, I have the same problem as you. Did you fix it?
> Thanks!

抱歉我有点记不清了 用 apt 试试看