floatlazer/semantic_slam #3: catkin_make error

amiltonwong, 2018-09-18:
Hi, @floatlazer ,

Thanks a lot for releasing your package. I had followed your install steps. However, I got error during catkin_make (detailed [log](https://gist.githubusercontent.com/amiltonwong/2a55b3145858fd0b3d4ccfc7ccea4ef3/raw/ef7046da2736e6d6685c859d99a2eda66d8b57e9/log))

It seems the issue related to building `octomap_generator`

```
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libvtkproj4-6.2.so.6.2.0', needed by '/root/catkin_ws/devel/lib/octomap_generator/octomap_generator'.  Stop.
CMakeFiles/Makefile2:2476: recipe for target 'semantic_slam/octomap_generator/CMakeFiles/octomap_generator.dir/all' failed
make[1]: *** [semantic_slam/octomap_generator/CMakeFiles/octomap_generator.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

Prior to executing `catkin_make`, I had executed `rosdep install semantic_slam` without any error.

Could you suggest me how to fix such install issue?

THX!

floatlazer, 2018-09-19:
Hi,

I did't encountered this issue during the work so I don't know how to solve this exactly. Apparently there is a library not found in your system. I think you shall fix the installation of libvtkproj4. Maybe [this issue](https://github.com/PointCloudLibrary/pcl/issues/1594) could help.

Cheers.

pranjal-mordor, 2020-03-13:
Hi @amiltonwong , were you able to solve this problem? If yes can you please share , as I am stuck with the same problem.

kmfrick, 2020-04-30:
If you have root access, this can probably be solved by `sudo apt install libvtk6.2` (assuming ROS Kinetic on Ubuntu Xenial).