floatlazer/semantic_slam #5: Python environment's problem

michaelczhou, 2018-10-12:
Hi:
Your job looks like awesome.When I run "roslaunch semantic_slam semantic_mapping.launch",I met some wrong.
```sh
[ INFO] [1539334068.008475645]: Semantic octomap generator [max fusion]
Traceback (most recent call last):
  File "/home/zc/catkin_car/src/semantic_slam-master/semantic_cloud/src/semantic_cloud.py", line 21, in <module>
    from color_pcl_generator import PointType, ColorPclGenerator
  File "/home/zc/catkin_car/devel/lib/python2.7/dist-packages/color_pcl_generator/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
ImportError: cannot import name 'PointType'
```

jsdd25, 2019-02-11:
@michaelczhou , Hi, I have the same problem as you. Did you fix it?
Thanks!

michaelczhou, 2019-02-11:
@jsdd25 make sure your python's environment is belong to ROS ,my mistake is forgetting to close annoconda. if this can't work ,you can try to reinstall ROS.

harnix, 2019-04-18:
@jsdd25 I ran into the same issue because I had multiple python versions installed. This was solved by specifying the right python version to run the code within semantic_cloud.py ros node. Hope that helps

zouyajing, 2019-10-31:
> @jsdd25 make sure your python's environment is belong to ROS ,my mistake is forgetting to close annoconda. if this can't work ,you can try to reinstall ROS.

Hi. I have both python2.7 and 3.5. I have tried two ways but both not work for me.
1. #!/usr/bin/env python2.7 or 3.5
2. catkin build --cmake-args -DPYTHON_VERSION=3.5
Can you tell me more details about solving this problem ?

zouyajing, 2019-10-31:
@harnix  Sorry. can you tell me more detail about avoiding this error ? I am a little confused about the python environment. Thanks.

AnhTuDo1998, 2021-01-07:
For whoever running into the same problem, pls try delete the content of the color_pcl_generator/__init__.py in the include folder (make it empty .py file)

reem90, 2021-01-20:
@AnhTuDo1990 What do you mean by empty? the content of the file or the name of the file? 
@all I am using python 3.8, I changed the first line in the semantic_labelling.py to be `#!/usr/bin/env python3.8 `and I got the following error 

`ImportError: cannot import name 'PointType' from partially initialized module 'color_pcl_generator' (most likely due to a circular import) (/home/hme/catkin_ws/devel/lib/python3/dist-packages/color_pcl_generator/__init__.py)`

did any one solved it?

AnhTuDo1998, 2021-01-21:
@reem90 make a __init__.py file with nothing inside

52nogamenolife, 2021-11-08:
@michaelczhou I have the same importerror , how did you solve it? I have tried to make an empty init.py, but i didn't work
 File "/home/lx/work/semantic_slam-master/slam/src/semantic_cloud/src/semantic_cloud.py", line 22, in <module>
    from color_pcl_generator import PointType, ColorPclGenerator
ImportError: cannot import name 'PointType'
what should I do? Many thanks!

russelldj, 2022-01-26:
I'm lacking context, but for future reference, I believe the correct file is `__init__.py` not `init.py`.