floatlazer/semantic_slam #6: Using Kinect V1 with openni binding

yaxauw, 2018-10-12:
It's really a pretty implementation of semantic SLAM!

However, I encounter an issue,

since I'm using Kinect V1 as my RGB-D camera, its default binding is with openni,

while the code here is binding with openni2,

`No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-kinetic-openni2-camera-0.4.0/src/openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.`

https://blog.csdn.net/sunyoop/article/details/78594871

the above link is the related issue, I'm wondering if I could use similar way to handle this problem,

to export something in .bashrc.

Hope to get your response,

Sincerely, Lambert

floatlazer, 2018-10-12:
Hi,  Lambert

Glad that you tried it out. Unfortunately I have really no experience with depth cameras other than asus xtion. So I am not able to fix your issue. But you may use other drivers available in ROS as long as image streamings outputs on right topics. Hope that you can figure it out.

Xuan

yaxauw, 2018-10-18:
Thank you for replying me! After several trying I want to try non-real time way, maybe it's more suitable for me due to the hardware issue. Just a short question, is that available to use rgb-d 2D images to output the same result? Thanks! 
Lambert

floatlazer, 2018-10-19:
You may try to stream color image to topic "/camera/rgb/image_raw" and depth image to topic "camera/depth_registered/image_raw".

If you want to don't want to use SLAM (i.e. launch only semantic_cloud.py & octomap_generator), you should also broadcast a tf transform from frame "camera_rgb_optical_frame" to "world" (child frame). 

Xuan

NicksonYap, 2018-12-10:
@floatlazer

Hi Xuan,
I'm not familiar with ROS, but have done a lot of software integration works.

What PUBSUB lobrary is used for publishing to these topics?

I'm considering integrating with other sensors, or even my own Multi View Stereo camera (two usb cameras)

wk524629918, 2021-01-29:
> @floatlazer
> 
> Hi Xuan,
> I also have a problem,
> I currently want to convert RGBD dataset to rosbag file,
> but I am not familiar with rosbag file, would you show me some suggestion?
> thanks!

Hi, have you solved this problem?

xiaoyuamw, 2021-07-14:
Hi, @hdtsai 

I want to know had you ever solved your issue and run successfully? If so, can you tell me how you did?
Thank you very much!

Best Regards.