floatlazer/semantic_slam #7: catkin_make error

csufangyu, 2018-10-27:
Thanks very much for your work,I try to run code .But I meet a problems,cloud you help me?? Thanks!
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:182:79:   required from here
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:22:20: error: ‘class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘getNodeChild’
     node->copyData(*(this->getNodeChild(node, 0)));
                    ^
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:29:7: error: ‘class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘deleteNodeChild’
       this->deleteNodeChild(node, i);
       ^
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx: In instantiation of ‘bool octomap::SemanticsOcTree<SEMANTICS>::isNodeCollapsible(const octomap::SemanticsOcTreeNode<SEMANTICS>*) const [with SEMANTICS = octomap::SemanticsMax]’:
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:182:79:   required from here
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:42:8: error: ‘const class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘nodeChildExists’
     if(!this->nodeChildExists(node, 0))
        ^
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:44:82: error: ‘const class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘getNodeChild’
     const SemanticsOcTreeNode<SEMANTICS>* firstChild = this->getNodeChild(node, 0);
                                                                                  ^
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:45:5: error: ‘const class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘nodeHasChildren’
     if(this->nodeHasChildren(firstChild))
     ^
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:51:43: error: ‘const class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘getNodeChild’
           || !(this->getNodeChild(node, i)->getSemantics() == firstChild->getSemantics()))
                                           ^
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:50:43: error: ‘const class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘getNodeChild’
           || !(this->getNodeChild(node, i)->getValue() == firstChild->getValue())
                                           ^
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:49:43: error: ‘const class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘nodeHasChildren’
       if (!this->nodeChildExists(node, i) || this->nodeHasChildren(this->getNodeChild(node, i))
                                           ^
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:49:43: error: ‘const class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘getNodeChild’
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:49:11: error: ‘const class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘nodeChildExists’
       if (!this->nodeChildExists(node, i) || this->nodeHasChildren(this->getNodeChild(node, i))
           ^
In file included from /opt/ros/indigo/include/octomap/OcTreeBaseImpl.h:503:0,
                 from /opt/ros/indigo/include/octomap/OccupancyOcTreeBase.h:44,
                 from /opt/ros/indigo/include/octomap/ColorOcTree.h:40,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.h:10,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/octomap_generator/octomap_generator.h:6,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:1:
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.hxx: In instantiation of ‘void octomap::OcTreeBaseImpl<NODE, INTERFACE>::pruneRecurs(NODE*, unsigned int, unsigned int, unsigned int&) [with NODE = octomap::SemanticsOcTreeNode<octomap::SemanticsBayesian>; INTERFACE = octomap::AbstractOccupancyOcTree]’:
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.hxx:409:51:   required from ‘void octomap::OcTreeBaseImpl<NODE, INTERFACE>::prune() [with NODE = octomap::SemanticsOcTreeNode<octomap::SemanticsBayesian>; INTERFACE = octomap::AbstractOccupancyOcTree]’
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:182:79:   required from here
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.hxx:590:72: error: invalid conversion from ‘octomap::ColorOcTreeNode*’ to ‘octomap::SemanticsOcTreeNode<octomap::SemanticsBayesian>*’ [-fpermissive]
           pruneRecurs(node->getChild(i), depth+1, max_depth, num_pruned);
                                                                        ^
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.hxx:582:8: error:   initializing argument 1 of ‘void octomap::OcTreeBaseImpl<NODE, INTERFACE>::pruneRecurs(NODE*, unsigned int, unsigned int, unsigned int&) [with NODE = octomap::SemanticsOcTreeNode<octomap::SemanticsBayesian>; INTERFACE = octomap::AbstractOccupancyOcTree]’ [-fpermissive]
   void OcTreeBaseImpl<NODE,I>::pruneRecurs(NODE* node, unsigned int depth,
        ^
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.hxx: In instantiation of ‘void octomap::OcTreeBaseImpl<NODE, INTERFACE>::expandRecurs(NODE*, unsigned int, unsigned int) [with NODE = octomap::SemanticsOcTreeNode<octomap::SemanticsBayesian>; INTERFACE = octomap::AbstractOccupancyOcTree]’:
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.hxx:418:38:   required from ‘void octomap::OcTreeBaseImpl<NODE, INTERFACE>::expand() [with NODE = octomap::SemanticsOcTreeNode<octomap::SemanticsBayesian>; INTERFACE = octomap::AbstractOccupancyOcTree]’
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:182:79:   required from here
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.hxx:623:59: error: invalid conversion from ‘octomap::ColorOcTreeNode*’ to ‘octomap::SemanticsOcTreeNode<octomap::SemanticsBayesian>*’ [-fpermissive]
         expandRecurs(node->getChild(i), depth+1, max_depth);
                                                           ^
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.hxx:607:8: error:   initializing argument 1 of ‘void octomap::OcTreeBaseImpl<NODE, INTERFACE>::expandRecurs(NODE*, unsigned int, unsigned int) [with NODE = octomap::SemanticsOcTreeNode<octomap::SemanticsBayesian>; INTERFACE = octomap::AbstractOccupancyOcTree]’ [-fpermissive]
   void OcTreeBaseImpl<NODE,I>::expandRecurs(NODE* node, unsigned int depth,
        ^
In file included from /opt/ros/indigo/include/octomap/OccupancyOcTreeBase.h:530:0,
                 from /opt/ros/indigo/include/octomap/ColorOcTree.h:40,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.h:10,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/octomap_generator/octomap_generator.h:6,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:1:
/opt/ros/indigo/include/octomap/OccupancyOcTreeBase.hxx: In instantiation of ‘std::ostream& octomap::OccupancyOcTreeBase<NODE>::writeBinaryNode(std::ostream&, const NODE*) const [with NODE = octomap::SemanticsOcTreeNode<octomap::SemanticsBayesian>; std::ostream = std::basic_ostream<char>]’:
/opt/ros/indigo/include/octomap/OccupancyOcTreeBase.hxx:958:7:   required from ‘std::ostream& octomap::OccupancyOcTreeBase<NODE>::writeBinaryData(std::ostream&) const [with NODE = octomap::SemanticsOcTreeNode<octomap::SemanticsBaye [... cut ...]
   void OcTreeBaseImpl<NODE,I>::calcNumNodesRecurs(NODE* node, size_t& num_nodes) const {
        ^
make[2]: *** [semantic_slam/octomap_generator/CMakeFiles/octomap_generator.dir/src/octomap_generator/octomap_generator.cpp.o] 错误 1
make[2]: *** 正在等待未完成的任务....
make[2]: *** [semantic_slam/octomap_generator/CMakeFiles/octomap_generator.dir/src/octomap_generator/octomap_generator_ros.cpp.o] 错误 1
make[1]: *** [semantic_slam/octomap_generator/CMakeFiles/octomap_generator.dir/all] 错误 2
make: *** [all] 错误 2
Invoking "make -j12 -l12" failed

floatlazer, 2018-10-28:
You didn't install ```octomap``` library properly. Try 
```sh
rosdep install semantic_slam
```

csufangyu, 2018-10-29:
I think it maybe related with the version of octomap. could you tell me the version of octomap you used! the more information ia as follows:
Scanning dependencies of target octomap_generator
[ 20%] Building CXX object semantic_slam/octomap_generator/CMakeFiles/octomap_generator.dir/src/octomap_generator/octomap_generator.cpp.o
[ 40%] Building CXX object semantic_slam/octomap_generator/CMakeFiles/octomap_generator.dir/src/semantic_octree/semantics_bayesian.cpp.o
[ 60%] Building CXX object semantic_slam/octomap_generator/CMakeFiles/octomap_generator.dir/src/octomap_generator/octomap_generator_ros.cpp.o
[ 80%] Building CXX object semantic_slam/octomap_generator/CMakeFiles/octomap_generator.dir/src/semantic_octree/semantics_max.cpp.o
In file included from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.h:11:0,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/octomap_generator/octomap_generator.h:6,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:1:
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree_node.h: In member function ‘void octomap::SemanticsOcTreeNode<SEMANTICS>::copyData(const octomap::SemanticsOcTreeNode<SEMANTICS>&)’:
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree_node.h:39:7: error: ‘copyData’ is not a member of ‘octomap::ColorOcTreeNode’
       ColorOcTreeNode::copyData(from);
       ^
In file included from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.h:11:0,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/octomap_generator/octomap_generator_ros.h:6,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator_ros.cpp:1:
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree_node.h: In member function ‘void octomap::SemanticsOcTreeNode<SEMANTICS>::copyData(const octomap::SemanticsOcTreeNode<SEMANTICS>&)’:
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree_node.h:39:7: error: ‘copyData’ is not a member of ‘octomap::ColorOcTreeNode’
       ColorOcTreeNode::copyData(from);
       ^
In file included from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/octomap_generator/octomap_generator.h:6:0,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:1:
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.h: In instantiation of ‘octomap::SemanticsOcTree<SEMANTICS>::StaticMemberInitializer::StaticMemberInitializer() [with SEMANTICS = octomap::SemanticsMax]’:
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:139:64:   required from ‘octomap::SemanticsOcTree<octomap::SemanticsMax>::StaticMemberInitializer octomap::SemanticsOcTree<octomap::SemanticsMax>::semanticsOcTreeMemberInit’
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:12:5:   required from ‘octomap::SemanticsOcTree<SEMANTICS>::SemanticsOcTree(double) [with SEMANTICS = octomap::SemanticsMax]’
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:11:103:   required from ‘OctomapGenerator<CLOUD, OCTREE>::OctomapGenerator() [with CLOUD = pcl::PointCloud<PointXYZRGBSemanticsMax>; OCTREE = octomap::SemanticsOcTree<octomap::SemanticsMax>]’
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:181:16:   required from here
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.h:97:12: error: ‘class octomap::SemanticsOcTree<octomap::SemanticsMax>’ has no member named ‘clearKeyRays’
            tree->clearKeyRays();
            ^
In file included from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.h:116:0,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/octomap_generator/octomap_generator.h:6,
                 from /home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/src/octomap_generator/octomap_generator.cpp:1:
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx: In instantiation of ‘void octomap::SemanticsOcTree<SEMANTICS>::updateInnerOccupancyRecurs(octomap::SemanticsOcTreeNode<SEMANTICS>*, unsigned int) [with SEMANTICS = octomap::SemanticsMax]’:
/home/robot/slam_project/catkin_ws/src/semantic_slam/octomap_generator/include/semantics_octree/semantics_octree.hxx:116:5:   required from ‘void octomap::SemanticsOcTree<SEMANTICS>::updateInnerOccupancy() [with SEMANTICS = octomap::SemanticsMax]’
[... cut ...]
           calcNumNodesRecurs(node->getChild(i), num_nodes);
                                                          ^
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.hxx:839:8: note: initializing argument 1 of ‘void octomap::OcTreeBaseImpl<NODE, INTERFACE>::calcNumNodesRecurs(NODE*, std::size_t&) const [with NODE = octomap::SemanticsOcTreeNode<octomap::SemanticsMax>; INTERFACE = octomap::AbstractOccupancyOcTree; std::size_t = long unsigned int]’
   void OcTreeBaseImpl<NODE,I>::calcNumNodesRecurs(NODE* node, size_t& num_nodes) const {