floatlazer/semantic_slam #11: Monocular Semantic Slam with ORB-SLAM2?

NicksonYap, 2018-12-10:
Hi,

ORB-SLAM2 supports monocular slam

I wonder if this code can be modified to use ORB-SLAM2's Monocular SLAM, instead of directly using RGB-D?

Thanks!

NicksonYap, 2018-12-10:
I'm not a pro, but Is actual scale really required for Semantic Slam or Semantic Segmentation in general?

Say that we give up on single camera,
Would multi camera (2 and above) help?

Or at least integration/support for lower cost / more widely used RGB-D sensors such as the Orbbec Astra Pro or Intel Realsense D435 (going for 180 USD now)

floatlazer, 2018-12-11:
Stereo camera could work as ORB-SLAM2 supports stereo cameras.

Semantic segmentation is done based on RGB image. Depth information is only required for SLAM and reconstruction.

We used Asus xtion camera in our experiments, other low cost cameras should also work.

NicksonYap, 2018-12-13:
@floatlazer 
I'm trying to buy the same sensor you're using.
What's the exact model name of the ones used by you?

There is the regular 
Asus Xtion,
Asus Xtion **PRO**
Asus Xtion **Live**
Asus Xtion **PRO Live**

All so confusing...

http://wiki.ipisoft.com/Depth_Sensors_Comparison#Xtion_Live_vs_Xtion_vs_Carmine
(See the most bottom part)

**PRO** seems mean for Developers (same hardware but different software?)
**Live** seems to mean it has an RGB sensor

Since RGB was used, I suppose you're using either the Asus Xtion **Live** or Asus Xtion **PRO Live**?

can you check if yours is "PRO" and whether it is needed (PRO is more costly)

NicksonYap, 2018-12-13:
https://github.com/floatlazer/semantic_slam/issues/12

Please

Neetika-Gupta, 2019-08-21:
@NicksonYap 
were you able to implement Monocular semantic SLAM?

NicksonYap, 2019-08-22:
@Neetika-Gupta 

Nope, did not give it a shot