floatlazer/semantic_slam #14: TF error

araster1010, 2019-01-04:
I'm trying to use your great work.

When I use the semantic_mapping.launch, the octomap_generator says below warning.

`[ WARN] [1546579727.548125655]: MessageFilter [target=/map ]:   The majority of dropped messages were due to messages growing older than the TF cache time.  The last message's timestamp was: 1546579726.635539, and the last frame_id was: /new
[ WARN] [1546579727.558286995]: MessageFilter [target=/map ]: Dropped 100.00% of messages so far. Please turn the [ros.octomap_generator.message_notifier] rosconsole logger to DEBUG for more information.
`

/map is the world coordinate. /new is my camera's coordinate. (I don't use ORB-SLAM)
Do you have any advice? Thanks.

araster1010, 2019-01-06:
Thanks. The warning disappeared.

closed 2019-01-06 by araster1010

bqm1111, 2025-03-01:
> Thanks. The warning disappeared.

Hi. I am having the same issue. Could you tell me how did you debug this issue?