floatlazer/semantic_slam #15: in ubuntu 18.04, using melodic ROS, can not install octomap_rviz_plugins

jxr041100, 2019-01-18:
when I run this code, I found the following error. 
[ INFO] [1547840814.108426029]: Semantic octomap generator [max fusion]
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ERROR] [1547840816.371270368]: PluginlibFactory: The plugin for class 'octomap_rviz_plugin/ColorOccupancyGrid' failed to load.  Error: According to the loaded plugin descriptions the class octomap_rviz_plugin/ColorOccupancyGrid with base class type rviz::Display does not exist. Declared types are  rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin_tutorials/Imu
Traceback (most recent call last):
  File "/home/shijia/jianfeng/semantic_slam/src/semantic_slam/semantic_cloud/src/semantic_cloud.py", line 25, in <module>
    from skimage.transform import resize
  File "/usr/lib/python2.7/dist-packages/skimage/transform/__init__.py", line 1, in <module>
    from .hough_transform import (hough_line, hough_line_peaks,
  File "/usr/lib/python2.7/dist-packages/skimage/transform/hough_transform.py", line 4, in <module>
    from ._hough_transform import (_hough_circle,
  File "skimage/transform/_hough_transform.pyx", line 13, in init skimage.transform._hough_transform (skimage/transform/_hough_transform.c:24803)
  File "/usr/lib/python2.7/dist-packages/skimage/draw/__init__.py", line 1, in <module>
    from .draw import (circle, ellipse, set_color, polygon_perimeter,
  File "/usr/lib/python2.7/dist-packages/skimage/draw/draw.py", line 4, in <module>
    from .._shared._geometry import polygon_clip
  File "/usr/lib/python2.7/dist-packages/skimage/_shared/_geometry.py", line 4, in <module>
    from matplotlib import path, transforms
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/__init__.py", line 130, in <module>
    from matplotlib.rcsetup import defaultParams, validate_backend, cycler
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/rcsetup.py", line 29, in <module>
    from matplotlib.fontconfig_pattern import parse_fontconfig_pattern
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/fontconfig_pattern.py", line 28, in <module>
    from backports.functools_lru_cache import lru_cache
ImportError: No module named functools_lru_cache
[semantic_cloud-2] process has died [pid 4610, exit code 1, cmd /home/shijia/jianfeng/semantic_slam/src/semantic_slam/semantic_cloud/src/semantic_cloud.py __name:=semantic_cloud __log:=/home/shijia/.ros/log/ccbacf12-1b59-11e9-88bc-f87588ade1fa/semantic_cloud-2.log].
log file: /home/shijia/.ros/log/ccbacf12-1b59-11e9-88bc-f87588ade1fa/semantic_cloud-2*.log

harnix, 2019-04-21:
Hi @jxr041100 you need to manually build from https://github.com/OctoMap/octomap_rviz_plugins and source the setup.bash for your current workspace.

liaoheqi123, 2019-07-10:
@harnix   hi，could you give some more details?

liaoheqi123, 2019-07-10:
hi@jxr041100  Did you solve this issue?

zhaozhongch, 2020-06-02:
You download the rviz_plugins and put the folder in your ros workspace/src
then `catkin_make` then the plugin will be installed

AnhTuDo1998, 2020-08-20:
I'm using Ros noetic with ubuntu 20.04 LTS and having the same issue. Putting the source code of octomap_rviz_pluggins into catkin_ws and build solved the problem.