floatlazer/semantic_slam #18: Kinect V2RGB-Dcamera

qxdaaaaa, 2019-04-24:
I try to use Kinect V2RGB-D camera ,but i don't konw what should i do to change your code,such as the ros topic,can you give me some suggestions?I want to construct  my own lab environmen slam 
Thanks

fishermeeeeen, 2019-10-15:
open ORB-SLAM2/Example/ROS/ORBSLAM2/src/ros_rgbd.cc   and change it : 
message_filters::Subscriber<sensor_msgs::Image> rgb_sub(nh,"/kinect2/qhd/image_color",1);
message_filters::Subscriber<sensor_msgs::Image> depth_sub(nh,"/kinect2/qhd/image_depth_rect",1);

then build and catkin_make

qxdaaaaa, 2019-11-23:
hello,have you replace the rersult?I can't make the semantic picture.may be I make the error change in rviz.

qxdaaaaa, 2019-11-23:
@floatlazer @fishermeeeeen

xiaoyuamw, 2021-07-14:
Hi, @qxdaaaaa 
I want to know had you ever solved your issue and run successfully? If so, can you tell me how you did?
Thank you very much!

Best Regards.