floatlazer/semantic_slam #19: how to run the TUM rgbd dataset?

Vivienfanghua57, 2019-04-24:
Thank you for your excellent work! 

I have run your code with demo.bag successfully. But when I tried the TUM rgbd dataset, the octomap visualization is just empty, with log:
[](url)
[image]
and the rviz is like this:
[image]

Is there anythins wrong with it?

utvc, 2020-01-15:
Hi,did you solved the problem I am facing the same problem.

pranjal-mordor, 2020-03-14:
@floatlazer Great work. However, I am also unable to run any other .bag files. How to solve this problem. @Vivienfanghua57  @utvc

wk524629918, 2021-01-29:
Have you solved the problem？