floatlazer/semantic_slam #20: Octomap_generator process has died. Not able to visualize the reconstruction in RViz

abhattce, 2019-04-29:
I get the following output when I run the rosplay  with demo.bag after running semantic_mapping.launch. What am I doing wrong? I have ubuntu 18 and ROS melodic.

NODES
  /
    octomap_generator (octomap_generator/octomap_generator)
    rviz (rviz/rviz)
    semantic_cloud (semantic_cloud/semantic_cloud.py)

ROS_MASTER_URI=http://localhost:11311

process[semantic_cloud-1]: started with pid [23221]
process[octomap_generator-2]: started with pid [23222]
process[rviz-3]: started with pid [23223]
[ INFO] [1556497430.446361309]: Semantic octomap generator [max fusion]
2
shutdown request: new node registered with same name
Generate semantic point cloud [bayesian fusion].
Setting up CNN model...
Setting up ROS...
Ready.
/home/adi/.local/lib/python2.7/site-packages/torch/nn/functional.py:2351: UserWarning: nn.functional.upsample is deprecated. Use nn.functional.interpolate instead.
  warnings.warn("nn.functional.upsample is deprecated. Use nn.functional.interpolate instead.")
[octomap_generator-2] process has died [pid 23222, exit code -11, cmd /home/adi/catkin3_ws/devel/lib/octomap_generator/octomap_generator __name:=octomap_generator __log:=/home/adi/.ros/log/c3aa01be-6a09-11e9-b670-d8fc936ee38f/octomap_generator-2.log].
log file: /home/adi/.ros/log/c3aa01be-6a09-11e9-b670-d8fc936ee38f/octomap_generator-2*.log

fishermeeeeen, 2019-10-15:
Hi,did you solved the problem I am facing the same problem.

qxdaaaaa, 2019-11-28:
Hi,did you solved the problem I am facing the same problem.

@fishermeeeeen @abhattce

gyl327, 2023-05-18:
Hi,did you solved the problem I am facing the same problem.
@floatlazer @fishermeeeeen @qxdaaaaa