floatlazer/semantic_slam #26: semantic_cloud.py unable to import from color_pcl_generator

AnhTuDo1998, 2020-08-20:
Hi, anybody know how to solve this?

Traceback (most recent call last):
  File "/home/boom/catkin_ws/src/semantic_slam/semantic_cloud/src/semantic_cloud.py", line 21, in <module>
    from color_pcl_generator import PointType, ColorPclGenerator
  File "/home/boom/catkin_ws/devel/lib/python3/dist-packages/color_pcl_generator/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
ImportError: cannot import name 'PointType' from partially initialized module 'color_pcl_generator' (most likely due to a circular import) (/home/boom/catkin_ws/devel/lib/python3/dist-packages/color_pcl_generator/__init__.py)
[semantic_cloud-2] process has died [pid 6982, exit code 1, cmd /home/boom/catkin_ws/src/semantic_slam/semantic_cloud/src/semantic_cloud.py __name:=semantic_cloud __log:=/home/boom/.ros/log/08006776-e2f5-11ea-a59a-6397d188ae3a/semantic_cloud-2.log].
log file: /home/boom/.ros/log/08006776-e2f5-11ea-a59a-6397d188ae3a/semantic_cloud-2*.log

I'm trying to run this with the demo rosbag and seems to be stuck at this issue.

closed 2020-09-23 by AnhTuDo1998

Yueting-Li, 2020-09-24:
Hi guy! have you solved this problem?

AnhTuDo1998, 2020-09-25:
@Yueting-Li which version of Python running? Maybe you can try modifying the __init__.py file in the include folder of color_pcl_generator.py and remove the import statement, seems to help for me

Yueting-Li, 2020-09-28:
Thanks for your advice! It seems to help for me, too. No error now.