floatlazer/semantic_slam #28: error about roslaunch semantic_mapping.launch

pugna111, 2020-08-31:
after roslaunch semantic_mapping.launch,the error imformation show in the terminal:
`[ERROR] [1598838171.042500]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7f8d2e0fe790>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 75, in callback
    self.signalMessage(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 57, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 291, in add
    self.signalMessage(*msgs)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 57, in signalMessage
    cb(*(msg + args))
  File "/home/titan/semantic_slam_ws/src/semantic_slam/semantic_cloud/src/semantic_cloud.py", line 210, in color_depth_callback
    cloud_ros = self.cloud_generator.generate_cloud_semantic_max(color_img, depth_img, semantic_color, pred_confidence, color_img_ros.header.stamp)
  File "/home/titan/semantic_slam_ws/src/semantic_slam/semantic_cloud/include/color_pcl_generator/color_pcl_generator.py", line 148, in generate_cloud_semantic_max
    self.generate_cloud_data_common(bgr_img, depth_img)
  File "/home/titan/semantic_slam_ws/src/semantic_slam/semantic_cloud/include/color_pcl_generator/color_pcl_generator.py", line 124, in generate_cloud_data_common
    self.bgr0_vect[:,0:1] = bgr_img[:,:,0].reshape(-1,1)
ValueError: could not broadcast input array from shape (921600,1) into shape (307200,1)
`
I'm using zed stereo camera,and I have changed color_image_topic and depth_image_topic in semantic_cloud.yaml,On the other hand  using own zed.yaml in ORBSLAM instead of xtion.yaml. cloud you give me some advices? thank you~

AnhTuDo1998, 2020-12-31:
Hi, what is the resolution of image from your camera?

frankSARU, 2021-07-28:
Hi there, I meet the same problem now and I'm using intel D435i camera. I never meet this error before, have you resolve your problem?

Gary-zyt, 2021-12-23:
Hi, I also meet the same problem now ，but I don't know how to correct errors, some people say maybe my numpy is too high ,but I try to reduced version, the problem remains .

dinojiang1003, 2024-02-28:
Hi, I also meet the same problem now，does any one solve this problem?