floatlazer/semantic_slam #31: ColorPclGenerator instance has no attribute 'generate_cloud'

mingjunlovelybaby, 2021-01-05:
This is A great job.  I am trying to run the color_pcl_generator.py file, but a little error occurred, how can I solve it? It seems that this function is not defined.
[image]
[image]

AnhTuDo1998, 2021-01-07:
Hi not sure if this work, but you can try to edit this following file such that it is just an empty __init__.py file
`semantic_cloud/include/color_pcl_generator/__init__.py`

AnhTuDo1998, 2021-01-07:
Might be related to this bunch of traps in Python:
http://python-notes.curiousefficiency.org/en/latest/python_concepts/import_traps.html