floatlazer/semantic_slam #33: ImportError: No module named rospkg

xhding1997, 2021-01-14:
I'm using Ubuntu 20.04 with noetic. I have already install full edition but I received a bug when I use this command:
roslaunch semantic_slam semantic mapping.launch

[ INFO] [1610618357.717406293]: Semantic octomap generator [max fusion]
Traceback (most recent call last):
  File "/home/bie/ros_ws/Sem_SLAM/src/semantic_slam/semantic_cloud/src/semantic_cloud.py", line 14, in <module>
    import rospy
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/__init__.py", line 49, in <module>
    from .client import spin, myargv, init_node, \
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 52, in <module>
    import roslib
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest  # noqa: F401
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
ImportError: No module named rospkg
[semantic_cloud-2] process has died [pid 28899, exit code 1, cmd /home/bie/ros_ws/Sem_SLAM/src/semantic_slam/semantic_cloud/src/semantic_cloud.py __name:=semantic_cloud __log:=/home/bie/.ros/log/2a1d8be0-564f-11eb-90c0-244bfecec228/semantic_cloud-2.log].
log file: /home/bie/.ros/log/2a1d8be0-564f-11eb-90c0-244bfecec228/semantic_cloud-2*.log

The rivz window was successfully opened, but the semantic_image shows "No Image".
I've tried so hard to fix this, but it didn't work.
Maybe I install anaconda3 and it change my terminal? But I've tried to vim ~/.bashrc and delete anaconda content, it didn't work.

AnhTuDo1998, 2021-01-15:
Are you using a catkin workspace for your project? Have you source the devel/setup.sh ?
If yes, can you try invoking the Python intepreter and try import rospkg?

xhding1997, 2021-03-07:
> Are you using a catkin workspace for your project? Have you source the devel/setup.sh ?
> If yes, can you try invoking the Python intepreter and try import rospkg?

It was because of python2.7 and python3.8. I don't install rospkg in python2.7, I change my default python
interpreter to 3.8 and slove this.
But I got a new trouble which I can't slove.
I try to run the demo.bag, but receive this:
`
/home/bie/.local/lib/python3.8/site-packages/torch/nn/functional.py:2952: UserWarning: nn.functional.upsample is deprecated. Use nn.functional.interpolate instead.
  warnings.warn("nn.functional.upsample is deprecated. Use nn.functional.interpolate instead.")
[ERROR] [1615104107.227323]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7fa7dd0d9910>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 76, in callback
    self.signalMessage(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 330, in add
    self.signalMessage(*msgs)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/home/bie/ros_ws/Sem_SLAM/src/semantic_slam/semantic_cloud/src/semantic_cloud.py", line 209, in color_depth_callback
    cloud_ros = self.cloud_generator.generate_cloud_semantic_max(color_img, depth_img, semantic_color, pred_confidence, color_img_ros.header.stamp)
  File "/home/bie/ros_ws/Sem_SLAM/src/semantic_slam/semantic_cloud/include/color_pcl_generator/color_pcl_generator.py", line 162, in generate_cloud_semantic_max
    return self.make_ros_cloud(stamp)
  File "/home/bie/ros_ws/Sem_SLAM/src/semantic_slam/semantic_cloud/include/color_pcl_generator/color_pcl_generator.py", line 138, in make_ros_cloud
    self.cloud_ros.data = np.getbuffer(memoryview(self.ros_data.ravel()))[:]
AttributeError: module 'numpy' has no attribute 'getbuffer'
`

I search a long time but get nothing to solve this. Looks like numpy cancel `numpy.getbuffer` function in python3, and use memoryview to alternative. But I don't know how to use memoryview in `color_pcl_generator.py` line 138. Can you help me with it ? I would really appreciate your help.