floatlazer/semantic_slam #38: semantic octomap using the rtabmap slam

germal, 2021-03-01:
Hi All,
I would use the slam of Rtabmap and the semantic data of semantic_slam to create a map
Do you find any solution to fuse rtabmap and semantic cloud , in order to feed the slam mapping data of rtabmap in the octomap generator ?
Thanks a lot in advance 
germal

FPSychotic, 2021-03-01:
Hehe, that would be good

germal, 2021-03-02:
...yes and I think also it would be possible 
Maybe it would need simply to modify the semantic_cloud.py with a transformation bgr->rgb in order to feed the segmented image it in rtabmap, but I don't know how to do it.Maybe someone that is good in OpenCV can do it ?
https://github.com/floatlazer/semantic_slam/blob/657814a1ba484de6b7f6f9d07c564566c8121f13/semantic_cloud/src/semantic_cloud.py#L57
Regards

frankSARU, 2021-09-16:
I'm trying this too. Is there any good method to use this semantic image on rtabmap?