floatlazer/semantic_slam #39: ImportError: cannot import name 'PointType' from partially initialized module 'color_pcl_generator'

xhding1997, 2021-03-04:
I try to use this command:  roslaunch semantic_slam semantic_mapping.launch
And I got some trouble here.

ImportError: cannot import name 'PointType' from partially initialized module 'color_pcl_generator' (most likely due to a circular import) (/home/bie/ros_ws/Sem_SLAM/devel/lib/python3/dist-packages/color_pcl_generator/__init__.py)
[semantic_cloud-2] process has died [pid 18524, exit code 1, cmd /home/bie/ros_ws/Sem_SLAM/src/semantic_slam/semantic_cloud/src/semantic_cloud.py __name:=semantic_cloud __log:=/home/bie/.ros/log/d5a89544-7cc5-11eb-a228-244bfecec228/semantic_cloud-2.log].
log file: /home/bie/.ros/log/d5a89544-7cc5-11eb-a228-244bfecec228/semantic_cloud-2*.log

A circular import ?
I'm using python3.8, does it matter?

xhding1997, 2021-03-04:
I have tried to set semantic_slam/semantic_cloud/include/color_pcl_generator/__init__.py to empty, but it doesn't work.

xhding1997, 2021-03-04:
I think I found some key problems.
When using the installation instructions recommended by the ROS official website (sudo apt-get install ros-noetic-desktop-full), Noetic will be installed to python3 at this time. You can find rospkg in the directory /usr/lib/python3/dist-packages.
But rospkg cannot be found in the directory /usr/lib/python2.7/dist-packages.
On another computer using ubuntu16.04, use sudo apt-get install ros-kinetic-desktop-full to install ros. You can find rospkg in the directory /usr/lib/python2.7/dist-packages.
When executing roslunch, the terminal will execute with python2.7, so rospkg cannot be found. When you change the terminal to the default python3 execution, you can find rospkg. But the new problem is that an error "most likely due to a circular import" will be reported. I have deleted the contents of __init__.py, and now it is a blank file, but it still cannot solve this bug.

Is it because this project cannot be run in the python3 environment?

ahmadkh1995, 2021-03-10:
Hi 
before run $ catkin_make just create a filename.txt file and insert below libraries and packagesthese to it and then : 

setuptools>=41.0.0
numpy>=1.15
scipy
Pillow
cython
opencv-python==3.3.1.11
matplotlib
scikit-image
tensorflow-gpu==1.13.1
keras==2.0.8
h5py
imageio==2.6.1
imgaug
pandas
future
torch
torchvision
protobuf
IPython[all]

just note than if you don't have gpu on your PC use " tensorflow==1.13.1 " instead.
and then install all of those packages and libraries with this command;

$  pip install filename.txt 

This will solve your problem.You don't to modify any source files of the project...

MarvinChung, 2021-12-17:
I have the same issue and I am using python3 and my rosdistro: noetic.
I solve this by changing the file semantic_slam/semantic_cloud/include/color_pcl_generator/init.py 
from 
```
from color_pcl_generator import PointType, ColorPclGenerator
```
to
```
from .color_pcl_generator import PointType, ColorPclGenerator
```

I think that using python2 and older ros version may not have this issue. Otherwise it might be that you forget to deactivate the conda environment which is shown in [issue 5: Python environment's problem](https://github.com/floatlazer/semantic_slam/issues/5)