floatlazer/semantic_slam #40: questions about the bag file

wk524629918, 2021-03-25:
I have successfully run demo.bag, I used the Kinect camera to record my own bag file, and the semantic segmentation is normal, but there is no semantic map. What should I pay attention to when making my own demo.bag?

ahmadkh1995, 2021-03-29:
Hi
For getting the semantic octomap in Rviz with your own dataset or camera , you have to modify "color_pcl_generator.py" and multiplicate below lines to Depth_Map_Factor , for example to 0.0001 ; 

        self.xyd_vect[:,0:2] = self.xy_index * depth_img.reshape(-1,1) / 1000
        self.xyd_vect[:,2:3] = depth_img.reshape(-1,1) / 1000

It's better you check this value and decrease or increase it.

Also check the camera intrinsic matrix parameter in this file "color_pcl_generator.py" and also parameter file (.yaml) corresponds to your camera's intrinsic matrix values , you can check it by;

       $ rostopic echo camera/rgb/camera_info

wk524629918, 2021-03-31:
> Hi
> For getting the semantic octomap in Rviz with your own dataset or camera , you have to modify "color_pcl_generator.py" and multiplicate below lines to Depth_Map_Factor , for example to 0.0001 ; [...]
Thank you for your reply. How do I get my own  Depth_Map_Factor?