floatlazer/semantic_slam #42: How to save the otcomap?

valar-JRJ, 2021-04-13:
I run the launch file semantic_mapping.launch successfully but the map didn't get saved. How to save the octomap?

AnhTuDo1998, 2021-04-14:
It should be automatically saved based on the path provided in the octomap_generator.yaml file. Try changing the path to "/home/your_user_name/map_name.ot" or "/home/your_user_name/map_name.bt"

valar-JRJ, 2021-04-29:
> It should be automatically saved based on the path provided in the octomap_generator.yaml file. Try changing the path to "/home/your_user_name/map_name.ot" or "/home/your_user_name/map_name.bt"

problem solved, thanks

closed 2021-04-29 by valar-JRJ

jason866695, 2025-01-20:
> It should be automatically saved based on the path provided in the octomap_generator.yaml file. Try changing the path to "/home/your_user_name/map_name.ot" or "/home/your_user_name/map_name.bt"

does that means,after i run the semantic_mapping.launch and finish the map building,and shut down the command windows,then i can find my map with my order path?because i cant find my map,and i dont know why,thanks.